bkerler/oppo_ozip_decrypt #4: decryption of .ozip and .ofp firmware files.

techmashido, 2019-03-14:
Dear sir,
I would like to convert my phone Realme 2 Pro firmware into QFIL/QPST flashable one (I saw some guides on internet so I want to try ) so that we can unbrick it in case of brick while flashing custom rom or doing any mistake in same. 

We have .ozip firmware files from company and .ofp file with encrypted msmdownloadtool from a Indonesian man who release firmware files for different devices on his site. 

Please, Can you tell me where can I find decryption keys so that I can provide to include into your scripts ? 

Regards

Jwaleed8, 2019-04-20:
Please Add RealMe U1 (RMX1831) ozip extract

bkerler, 2019-04-23:
can you send me the "/vendor/lib64/libapplypatch_jni.so" for that device ?

Jwaleed8, 2019-04-23:
How to get it without root?

Jwaleed8, 2019-04-23:
https://github.com/realme-kernel-opensource/RM1_AND_RMU1-kernel-source

Realme U1 (RMX1831) Device kernel Source

Jwaleed8, 2019-04-23:
Here's The ozip Firmware

https://download.c.realme.com/osupdate/RMX1831EX_11_OTA_0070_all_UqwwgT6ye4J1.ozip

bkerler, 2019-04-23:
adb pull isn't possible ? neither kernel source nor ozip will help

bkerler, 2019-04-24:
Done. See full tutorial for it over here : https://bkerler.github.io/2019/04/the-game-begins/

closed 2019-04-24 by bkerler

vishal12345678901234567890, 2021-01-02:
I want to convert my oppo f15 ozip to zip format please help me

Facimilies, 2021-07-31:
hello, i have a realme x7 pro and need a flashable file. i have a ozip file but i need to be able to flash via sp flash tools, how can i get a scatter file ? or a OFP to convert to scatter file from a ozip firmware ? please any help would be appreciated i have hard bricked my phone. ozip link for correct firmware attached

RMX2121_11_OTA_0170_all_IrbQNx72829t.ozip ( please note the RMX2121_PU_11 is wrong firmware and cannot be flashed to my phone )

https://droidfilehost.com/download/realme-x7-pro-rmx2121-firmware-flash-file-stock-rom/?ind=1615834432477&filename=RMX2121_11_OTA_0190_all_B5cSIYs99RcD.ozip&wpdmdl=4242&refresh=6104fb82697e11627716482

lakshywihdew, 2021-08-02:
Hello Please help i flashed a rom which was not compatible with my device... and now i cant flash any rom in twrp or any other recovery my device is realme 7 pro and i need to convert it into ofp file its zip right now