bkerler/oppo_ozip_decrypt #5: Problem decrypting ozip

hbarsaiyan, 2019-05-02:
My device is Realme 3 pro. I was trying to decrypt ozip (https://download.c.realme.com/osupdate/RMX1851EX_11_OTA_0120_all_L6ZahEviNKID.ozip) but /vendor/lib64/libapplypatch_jni.so does not exist in the device.

List of files in /vendor

/vendor:
app
bin
bt_firmware
dsp
etc
firmware_mnt
lib
lib64
lost+found
oppo
overlay
radio
rfs

/vendor/app:
ConnectionSecurityService
PowerOffAlarm
RfToolkit
SSGTelemetryService
TimeService
TrustZoneAccessService

/vendor/app/ConnectionSecurityService:
ConnectionSecurityService.apk

/vendor/app/PowerOffAlarm:
PowerOffAlarm.apk

/vendor/app/RfToolkit:
RfToolkit.apk
lib

/vendor/app/RfToolkit/lib:
arm64

/vendor/app/RfToolkit/lib/arm64:
libnativehelpers.so
liboddi.so
librftoolkit.so

/vendor/app/SSGTelemetryService:
SSGTelemetryService.apk

/vendor/app/TimeService:
TimeService.apk

/vendor/app/TrustZoneAccessService:
TrustZoneAccessService.apk

/vendor/bin:
acpi
base64
basename
blockdev
cal
cat
chcon
chgrp
chmod
chown
chroot
chrt
cksum
clear
cmp
[... 1170 lines cut ...]
ramdumps
readonly
readwrite
shared

/vendor/rfs/msm/mpss/readonly:
firmware
vendor

/vendor/rfs/msm/mpss/readonly/vendor:
firmware

/vendor/rfs/msm/slpi:
hlos
ramdumps
readonly
readwrite
shared

/vendor/rfs/msm/slpi/readonly:
firmware
vendor

/vendor/rfs/msm/slpi/readonly/vendor:
firmware

redfeast, 2019-05-02:
use adb to pull it, it exists but permissiom will be denied to list using apps in phone

redfeast, 2019-05-02:
also send me the file please, once you pull it, to redfeastred@gmail.com

hbarsaiyan, 2019-05-02:
adb pull /vendor/lib64/libapplypatch_jni.so
remote object '/vendor/lib64/libapplypatch_jni.so' does not exist

redfeast, 2019-05-02:
you have telegram?? gib id

hbarsaiyan, 2019-05-02:
@americanpsychopath

fawazahmed0, 2019-05-06:
@redfeast @hbarsaiyan Try using this:
https://drive.google.com/file/d/1DTipcIiASNNzB37hrRFLtwJ96aHjKTOy/view

bkerler, 2019-06-17:
Extract /sbin/recovery from recovery.img if libapplypatch_jni.so doesn't exist.

closed 2019-06-17 by bkerler