bkerler/oppo_ozip_decrypt #9: can't extract recovery from Reno OFP

roytam1, 2019-06-03:
running `ofp_libextract.py CPH1917EX_11_A.05_190413_9df33f2c.ofp` gives a corrupted recovery.cpio.gz

zhixiang520, 2019-06-03:
I can extract recovery.cpio.gz, but I am following the ghidra step.  Can
you communicate?

bkerler, 2019-06-17:
If you can share a link to the ofp file, I can have a look.

roytam1, 2019-06-17:
> If you can share a link to the ofp file, I can have a look.

http://www.mediafire.com/file/3n2ze3esc1qkqc3

roytam1, 2019-06-17:
as a note: rev 87f0464 works with Reno OFP now.

roytam1, 2019-06-17:
just extracted the key, but when I tried to decrypt Reno's OTA ozip, it raised an error:

````
E:\test\oppo_ozip_decrypt-master>c:\Python37-32\python.exe ozipdecrypt.py CPH1917EX_11_OTA_0080_all_GWbhnkTy2RvI.ozip
ozipdecrypt 0.4 (c) B.Kerler 2017-2019
Found correct AES key: ACAC1E13A12531AE4A1B21BB31C13C21
Decrypting...
Traceback (most recent call last):
  File "ozipdecrypt.py", line 97, in <module>
    main()
  File "ozipdecrypt.py", line 77, in main
    wf.write(ctx.decrypt(data))
  File "c:\Python37-32\lib\site-packages\Crypto\Cipher\_mode_ecb.py", line 195, in decrypt
    raise ValueError("Data must be aligned to block boundary in ECB mode")
ValueError: Data must be aligned to block boundary in ECB mode
````
ozip file size is 3,008,799,799 bytes.
dec file size is 3,008,790,528 bytes.

roytam1, 2019-06-17:
It seems that it uses different file structure for new systems: https://oppo-au.custhelp.com/app/software_update_detail/p_name/Reno

it seems to have standard zip file structure with encrypted data

closed 2019-06-17 by roytam1