bkerler/oppo_ozip_decrypt #10: add Realme C2 aes key / magic UnicodeDecodeError

aIecxs, 2019-06-10:
sorry i have a usage problem..

(regardless i have no aes key, yet) i do not understand how to decrypt ozip file. unfortunately i don't understand python, but from the source code i can see Magic OPPOENCRYPT! is expected.

the firmware i downloaded from http://www.tuserhp.com/2019/05/firmware-realme-c2-rmx1941.html is a ozip file. but i cannot see the Magic. instead, i can unpack the ozip file with p7zip. the files inside have Magic OPPOENCRYPT!

is the usage: "ozipdecrypt.py [*.ozip]" still the same, or should i run decryption on each file instead?

is the UnicodeDecodeError caused by wrong input or broken environment? i have found only this solution but when i change the script, i get "ValueError: binary mode doesn't take an encoding argument"
https://stackoverflow.com/questions/46180610/python-3-unicodedecodeerror-how-do-i-debug-unicodedecodeerror

**Here is the error:** `$ python ozipdecrypt.py RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip`
<details>
  <summary>Click to expand</summary>

`mint@debian ~/src/python-3.7.3 $ python3.7 -V`
`Python 3.7.3`
`mint@debian ~/src/python-3.7.3 $ python3.7 -c "import ssl; print(ssl.OPENSSL_VERSION)"`
`OpenSSL 3.0.0-dev xx XXX xxxx`
`mint@debian ~/src/python-3.7.3 $ sudo pip3.7 install pycrypto`
`Collecting pycrypto`
`  Downloading https://files.pythonhosted.org/packages/60/db/645aa9af249f059cc3a368b118de33889219e0362141e75d4eaf6f80f163/pycrypto-2.6.1.tar.gz (446kB)`
`     |########################################| 450kB 801kB/s`
`Installing collected packages: pycrypto`
`  Running setup.py install for pycrypto ... done`
`Successfully installed pycrypto-2.6.1`
`mint@debian ~/src/python-3.7.3 $ lsb_release -a`
`No LSB modules are available.`
`Distributor ID:	LinuxMint`
`Description:	LMDE 2 Betsy`
`Release:	2`
`Codename:	betsy`
`mint@debian ~/src/python-3.7.3 $ cd ..`
`mint@debian ~/src $ cd oppo_ozip_decrypt-master`
`mint@debian ~/src/oppo_ozip_decrypt-master $ ls`
`ofp_libextract.py  ozipdecrypt.py  README.md`
`mint@debian ~/src/oppo_ozip_decrypt-master $ python3.7 ozipdecrypt.py /media/mint/5E7B7301488C5437/Downloads/RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip`
`ozipdecrypt 0.3 (c) B.Kerler 2017-2019`
`Traceback (most recent call last):`
`  File "ozipdecrypt.py", line 74, in <module>`
`    main()`
`  File "ozipdecrypt.py", line 48, in main`
`    magic=fr.read(12).decode()`
`UnicodeDecodeError: 'utf-8' codec can't decode bytes in position 10-11: unexpected end of data`
`mint@debian ~/src/oppo_ozip_decrypt-master $ hexdump -C -n512 /media/mint/5E7B7301488C5437/Downloads/RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip`
`00000000  50 4b 03 04 0a 00 00 08  00 00 ec 8c 5b 4d 87 ff  |PK..........[M..|`
`00000010  70 88 ad 1a 00 00 ad 1a  00 00 11 00 0a 00 63 6f  |p.............co|`
`00000020  6d 70 61 74 69 62 69 6c  69 74 79 2e 7a 69 70 fe  |mpatibility.zip.|`
`00000030  ca 00 00 35 d9 02 00 00  00 50 4b 03 04 14 00 00  |...5.....PK.....|`
`00000040  00 08 00 00 00 21 3a 6c  a1 f8 45 99 07 00 00 aa  |.....!:l..E.....|`
`00000050  5b 00 00 13 00 00 00 76  65 6e 64 6f 72 5f 6d 61  |[......vendor_ma|`
`00000060  6e 69 66 65 73 74 2e 78  6d 6c ed 9c 5d 6f db 36  |nifest.xml..]o.6|`
`00000070  14 86 ef fb 2b 02 df 0e  b3 62 a7 18 86 c2 09 96  |....+....b......|`
`00000080  ba 49 53 b4 69 b3 26 5b  2f 03 5a a2 25 c2 92 a8  |.IS.i.&[/.Z.%...|`
`00000090  92 b4 b3 fc fb 51 5f 96  65 51 16 45 1f 5a 18 b0  |.....Q_.eQ.E.Z..|`
`000000a0  de b4 b5 79 f8 1e 3d 3c  fc 3e d6 2c 42 31 59 62  |...y..=<.>.,B1Yb|`
`000000b0  2e ce 36 98 71 42 e3 cb  d1 64 7c 3e 3a 13 af 09  |..6.qB...d|>:...|`
`000000c0  be 1c 79 78 43 5c 2c ff  87 98 8f c5 af 21 de e0  |..yxC\,......!..|`
`000000d0  f0 72 74 31 ba 7a 73 26  ff cc 02 14 9e 2d 29 8b  |.rt1.zs&.....-).|`
`000000e0  90 b8 1c 05 c4 0b 8b cf  b3 ef 62 14 e1 2b 14 7b  |..........b..+.{|`
`000000f0  8c 12 6f 1c 20 e6 bd 20  86 c7 68 ed 11 3a 73 b2  |..o. .. ..h..:s.|`
`00000100  2f ab b2 82 a1 98 27 94  89 ab e0 65 41 62 0f b3  |/.....'....eAb..|`
`00000110  99 53 7d 56 95 2b 5c bc  7a 3b 3e 9f 39 e5 7f aa  |.S}V.+\.z;>.9...|`
`00000120  af 49 2c 30 5b 22 77 a7  ea ca 95 4f 1f b2 67 e1  |.I,0["w....O..g.|`
`00000130  b7 c8 15 94 bd ee fb 50  54 c0 05 8a a5 bd 87 97  |.......PT.......|`
`00000140  68 1d 8a 99 b3 fd a4 52  71 14 32 b3 e5 cf ac ba  |h......Rq.2.....|`
`00000150  3f a4 67 ef de ed 49 39  db da 8a 52 39 3c 47 d2  |?.g...I9...R9<G.|`
`00000160  3b 8e e3 18 2f 97 d8 15  c3 e0 bc c9 b4 4f 82 b3  |;.../........O..|`
`00000170  2e 65 03 e7 22 5c 63 41  a9 08 8e 66 39 31 61 f9  |.e.."\cA...f91a.|`
`00000180  be 94 bf 73 89 25 92 93  8c e4 ae 90 0d 8e ae fc  |...s.%..........|`
`00000190  94 a1 71 c2 e8 86 64 a8  8e a4 39 1d bf ed 4f 73  |..q...d...9...Os|`
`000001a0  9e 39 f1 d0 e2 43 9d 67  56 53 8c 42 e7 bc 3f 52  |.9...C.gVS.B..?R|`
`000001b0  e9 9c 44 5a 57 73 76 2b  84 e4 ca 87 89 cc 7b ec  |..DZWsv+......{.|`
`000001c0  11 34 47 fc 11 b3 74 48  b3 1a 9c 7b 5a 56 e2 93  |.4G...tH...{ZV..|`
`000001d0  c6 4b e2 73 39 8c 34 1e  c5 80 e7 a4 3f cf c7 75  |.K.s9.4.....?..u|`
`000001e0  f6 dd 6d 48 62 1f b3 79  ee 8e 35 aa 13 49 55 a9  |..mHb..y..5..IU.|`
`000001f0  68 83 ad c7 a2 61 62 74  ce 5e 13 41 ed 4c 44 1d  |h....abt.^.A.LD.|`
`00000200`
`mint@debian ~/src/oppo_ozip_decrypt-master $ 7z l /media/mint/5E7B7301488C5437/Downloads/RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip`
` `
`7-Zip [32] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21`
`p7zip Version 16.02 (locale=en_US.UTF-8,Utf16=on,HugeFiles=on,32 bits,2 CPUs Intel(R) Core(TM)2 Duo CPU     E8400  @ 3.00GHz (10676),ASM)`
` `
`Scanning the drive for archives:`
`1 file, 2026637673 bytes (1933 MiB)`
` `
`Listing archive: /media/mint/5E7B7301488C5437/Downloads/RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip`
` `
`--`
`Path = /media/mint/5E7B7301488C5437/Downloads/RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip`
`Type = zip`
`Physical Size = 2026637673`
`Comment = signed by SignApk`
` `
`   Date      Time    Attr         Size   Compressed  Name`
`------------------- ----- ------------ ------------  ------------------------`
`2018-10-27 17:39:24 .....         6829         6829  compatibility.zip`
`2018-10-27 17:39:24 .....            0            0  system.patch.dat`
`2018-10-27 17:39:24 .....            1            1  type.txt`
`2018-10-27 17:39:24 .....            0            0  vendor.patch.dat`
`2018-10-27 17:39:24 .....          289          184  META-INF/com/android/metadata`
`2018-10-27 17:39:24 .....      1817384       846592  META-INF/com/google/android/update-binary`
`2018-10-27 17:39:24 .....         5531         1297  META-INF/com/google/android/updater-script`
`2018-10-27 17:39:24 .....     33558608      9462919  boot.img`
`2018-10-27 17:39:24 .....     16781392        69153  dtbo.img`
`2018-10-27 17:39:24 .....      1010384       667217  lk.img`
`2018-10-27 17:39:24 .....     50120000     40330397  md1img.img`
`2018-10-27 17:39:24 .....          175           92  oppo_metadata`
`2018-10-27 17:39:24 .....         1200          471  oppo_update_script.sh`
`2018-10-27 17:39:24 .....       251760       238368  preloader.img`
`2018-10-27 17:39:24 .....       251760       238368  preloader_emmc.img`
`2018-10-27 17:39:24 .....       253808       238377  preloader_ufs.img`
`2018-10-27 17:39:24 .....          777          328  scatter.txt`
`2018-10-27 17:39:24 .....       560640       183660  scp.img`
`2018-10-27 17:39:24 .....        56256        33232  spmfw.img`
`2018-10-27 17:39:24 .....       435072       429458  sspm.img`
`2018-10-27 17:39:24 .....   1664413241   1660041841  system.new.dat.br`
`2018-10-27 17:39:24 .....        21968        18150  system.transfer.list`
`2018-10-27 17:39:24 .....       482096       430718  tee.img`
`2018-10-27 17:39:24 .....         8272         3271  vbmeta.img`
`2018-10-27 17:39:24 .....    313724176    313383587  vendor.new.dat.br`
`2018-10-27 17:39:24 .....         7664         3846  vendor.transfer.list`
`2018-10-27 17:39:24 .....         1424          988  META-INF/com/android/otacert`
`2018-10-27 17:39:24 .....         1928          907  META-INF/MANIFEST.MF`
`2018-10-27 17:39:24 .....         2224          984  META-INF/CERT.SF`
`2018-10-27 17:39:24 .....         1522         1138  META-INF/CERT.RSA`
`------------------- ----- ------------ ------------  ------------------------`
`2018-10-27 17:39:24         2083776381   2026632373  30 files`
`mint@debian ~/src/oppo_ozip_decrypt-master $ python3.7 ozipdecrypt.py /media/mint/5E7B7301488C5437/Downloads/RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip`
`ozipdecrypt 0.3 (c) B.Kerler 2017-2019`
`Traceback (most recent call last):`
`  File "ozipdecrypt.py", line 74, in <module>`
`    main()`
`  File "ozipdecrypt.py", line 48, in main`
`    magic=fr.read(12).decode()`
`UnicodeDecodeError: 'utf-8' codec can't decode bytes in position 10-11: unexpected end of data`
`mint@debian ~/src/oppo_ozip_decrypt-master $`
</details>

also, i would ask you if it is possible to add the Realme C2 after i provided the libapplypatch_jni.so

aIecxs, 2019-06-16:
seems there is no libapplypatch_jni.so (adb pull failed)
/vendor/lib64
/vendor/lib
/system/lib64
/system/lib 

but i have found another firmware [RMX1941EX_11_A.10_190507_c80b7906.ofp](https://androidfilehost.com/?fid=6006931924117882343)
and your script successfully extracted [/sbin/recovery](https://www.mediafire.com/file/2s5jgi4xrx6eonu/sbin-recovery_RMX1941EX_11_A.10_190507_c80b7906.zip) from this!

Now i am a little lost. I wonder if i need to extract the libraries from rom too because /sbin/recovery shows a lot of unresolved dependencies. It is very hard for a noob to follow your tutorial, i do not understand how to trace back xrefs, double clicking on them does nothing?

can you please show where exactly to click and which xref is the right one?
[Screenshot from 2019-06-15 22-13-07.png](https://www.dropbox.com/s/ob0l6xpugjn1grj/Screenshot%20from%202019-06-15%2022-13-07.png?dl=0&raw=1)
[Screenshot from 2019-06-15 22-35-42.png](https://www.dropbox.com/s/v86713rv5oe8yhs/Screenshot%20from%202019-06-15%2022-35-42.png?dl=0&raw=1)

or at least, update the script with key for Realme C2
(and maybe new decryption/extraction method for these new? [kind of ozip](http://www.mediafire.com/file/ty41r311xvy6mok/RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip) files)?
the goal is to decrypt this [boot.img OPPOENCRYPT!](https://www.mediafire.com/file/gbfays9aaiwfd8c/boot-img_RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.zip) i have extracted from the previous indonesian link

i am currently trying to port a custom recovery based on the extracted recovery.cpio.gz maybe i can get the boot.img once obtained root root access

aIecxs, 2019-06-16:
local_4210 = 0xae3125a7131eacac;
  uVar5 = AES_set_decrypt_key((uchar *)&local_4210,0x80,&AStack352);

when i double click on uchar, then it jumps to 0x001156d4 (and next line is the pointer)

        001156d4 e0 43 00 91     add        x0,sp,#0x10
        001156d8 00 01 c0 3d     ldr        q0,[x8]=>DAT_001edd30

double click on DAT_001edd30, then it jumps to 0x001edd30

                             DAT_001edd30
        001edd30 ac              ??         ACh
        001edd31 ac              ??         ACh
        001edd32 1e              ??         1Eh
        001edd33 13              ??         13h
        001edd34 a7              ??         A7h
        001edd35 25              ??         25h
        001edd36 31              ??         31h
        001edd37 ae              ??         AEh
        001edd38 4a              ??         4Ah
        001edd39 1b              ??         1Bh
        001edd3a 22              ??         22h
        001edd3b bb              ??         BBh
        001edd3c 31              ??         31h
        001edd3d c1              ??         C1h
        001edd3e cc              ??         CCh
        001edd3f 22              ??         22h

so is this the key? 0xacac1e13a72531ae4a1b22bb31c1cc22
but this key is already in script for Realme 3, but i can not decrypt boot.img from Realme C2

or do i have to bitwise AND (&) calculate this with 0xae3125a7131eacac

10101100101011000001111000010011101001110010010100110001101011100100101000011011001000101011101100110001110000011100110000100010
&
1010111000110001001001011010011100010011000111101010110010101100
=
0000101000010001001000001010001100010001000000001000110000100000

= but this gives me 8 bytes key only 0x0a1120a311008c20
so is this the key? 0x00000000000000000a1120a311008c20

or should i just bitwise AND (&) calculate the last 8 bytes?
then i get this key, but still not able to decrypt 0xacac1e13a72531ae0a1120a311008c20

bkerler, 2019-06-17:
Actually, I found two keys :
ACAC1E13A72531AE4A1B22BB31C1CC22 (decryption)
2132321EA2CA86621A11241ABA512722 (encryption)

bkerler, 2019-06-17:
The aes test only was made for full images. I added boot.img support. Decrypts now fine.

closed 2019-06-17 by bkerler