bkerler/oppo_ozip_decrypt #13: Can't decrypt Reno OZIP

roytam1, 2019-06-17:
just extracted the key, but when I tried to decrypt Reno's OTA ozip, it raised an error:

````
E:\test\oppo_ozip_decrypt-master>c:\Python37-32\python.exe ozipdecrypt.py CPH1917EX_11_OTA_0080_all_GWbhnkTy2RvI.ozip
ozipdecrypt 0.4 (c) B.Kerler 2017-2019
Found correct AES key: ACAC1E13A12531AE4A1B21BB31C13C21
Decrypting...
Traceback (most recent call last):
  File "ozipdecrypt.py", line 97, in <module>
    main()
  File "ozipdecrypt.py", line 77, in main
    wf.write(ctx.decrypt(data))
  File "c:\Python37-32\lib\site-packages\Crypto\Cipher\_mode_ecb.py", line 195, in decrypt
    raise ValueError("Data must be aligned to block boundary in ECB mode")
ValueError: Data must be aligned to block boundary in ECB mode
````
ozip file size is 3,008,799,799 bytes.
dec file size is 3,008,790,528 bytes.

It seems that it uses different file structure for new systems: https://oppo-au.custhelp.com/app/software_update_detail/p_name/Reno

it seems to have standard zip file structure with encrypted data.

( tag #9  )

bkerler, 2019-06-20:
yes, will add the new structure soon

aIecxs, 2019-06-21:
@bkerler thanks for adding Realme C2

what about new decryption/extraction method for these new [kind of ozip](http://www.mediafire.com/file/ty41r311xvy6mok/RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip) files?

tried renaming to boot.img but does not work

boot.img .. OK
dtbo.img
lk.img
md1img.img
preloader.img
preloader_emmc.img
preloader_ufs.img
scp.img
spmfw.img
sspm.img
system.transfer.list
tee.img
vbmeta.img
vendor.transfer.list

it would be nice to have the whole ozip archive extracted at once, without knowing the file names. i guess this is related to this issue and you already working on it, just post it here to make sure

bkerler, 2019-06-21:
Solved with commit 844e75d

closed 2019-06-21 by bkerler

roytam1, 2019-06-22:
> Solved with commit [844e75d](https://github.com/bkerler/oppo_ozip_decrypt/commit/844e75d0369997ef94e826eec4e8a613653cd4e8)

still having issue in windows:

````E:\test\oppo_ozip_decrypt-master>c:\Python37-32\python.exe ozipdecrypt.py RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip
ozipdecrypt 0.5 (c) B.Kerler 2017-2019
Extracting RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip
Found correct AES key: acac1e13a72531ae4a1b22bb31c1cc22
Traceback (most recent call last):
  File "ozipdecrypt.py", line 151, in <module>
    main()
  File "ozipdecrypt.py", line 123, in main
    with open(wfilename,'wb') as wf:
FileNotFoundError: [Errno 2] No such file or directory: 'out/temp\\boot.img'
````

bkerler, 2019-06-22:
oh, sorry I usually use linux. Just added a fix which should solve your issue.

Jazeelahammed, 2020-05-22:
> > Solved with commit [844e75d](https://github.com/bkerler/oppo_ozip_decrypt/commit/844e75d0369997ef94e826eec4e8a613653cd4e8)
> 
> still having issue in windows:
> 
> ```
> ozipdecrypt 0.5 (c) B.Kerler 2017-2019
> Extracting RMX1941EX_11_OTA_0100_all_6Tw0LV9IfSI3.ozip
> Found correct AES key: acac1e13a72531ae4a1b22bb31c1cc22
> Traceback (most recent call last):
>   File "ozipdecrypt.py", line 151, in <module>
>     main()
>   File "ozipdecrypt.py", line 123, in main
>     with open(wfilename,'wb') as wf:
> FileNotFoundError: [Errno 2] No such file or directory: 'out/temp\\boot.img'
> ```
did you find solution please inform.....