bkerler/oppo_ozip_decrypt #16: Failed to find aes key for Reno Z (CPH1979 or PCDM10)

tsyu, 2019-08-15:
I tried to find aes key, using "/sbin/recovery" from extracted cpio.
(Btw, I can't find /vendor/lib64/libapplypatch_jni.so or anything like it in file system)

But it looks oppo no longer hard-coded 'realkey' while using AES_set_decrypt_key.
Any thoughts on how to trace this down?
Here's the /sbin/recovery file:
[recovery.zip](https://github.com/bkerler/oppo_ozip_decrypt/files/3504777/recovery.zip)

tsyu, 2019-08-15:
Sorry I made a mistake.  I saw this in ghidra output:
  local_4210 = 0xae3124a7131eacac;
  uVar5 = AES_set_decrypt_key((uchar *)&local_4210,0x80,&AStack352);

Since it's not 128-bit, I ignored this one.
Tracing local_4210 down I found the correct key: acac1e13a72431ae4a1b22bba1c1c6a2
I think ghidra translate it wrong.

chan1108, 2019-08-21:
If I want to find the aes key to decrypt ozip, should I have to find the ofp file? However, where can I get such a file?

ganeshi4u, 2019-10-20:
Hey @tsyu i have the same function with the local_4210 8 bit hex value, can you tell me how were you able to trace down local_4210? As for me i don't see any references for this variable other than within the function.

```
  uStack16888 = 0x21315131bb531b54;
  local_4200 = 0xae1325a2a3114a1c;
  iVar6 = AES_set_decrypt_key((uchar *)&local_4200,0x80,&AStack336)
```;

ganeshi4u, 2019-10-20:
ok nvm i have actually figured it out. I'll leave a few points here that might help someone in the future.

**Follow these simple steps:**
After finding the **aes_set_decrypt_key** function double click on the (uchar *) and it jumps to a address in the Listing view window. Now find a nearest address that has a pointer in it. Double click on that pointer and it'll jump to the 16bit key (vertical list).

closed 2019-11-25 by tsyu