bkerler/oppo_ozip_decrypt #27: Realme XT and ofp_libextract.py

gourab4u, 2019-11-09:
I tried to extract libapplypatch_jni.so with ofp_libextract.py however I did not get any libapplypatch_jni.so. On successful run, I got 108 recovery.cpio.gz (recovery1.cpio.gz - recovery108.cpio.gz)

one fine is of 24.2 MB, 13 files between 1- 10 MB, rest in KBs. I have checked in Ghidra but unable to get aes_decrypt functions in any of the file sizes in MBs. Am I doing anything wrong?

bkerler, 2020-01-01:
that's quite a lot. but you should have a look into the .gz files if there are the required files for reversing the key. see my blog for further information.

bkerler, 2020-02-01:
key has already been added. in the largest one (24.2), just extract sbin/recovery. Key is in there.

closed 2020-02-01 by bkerler