bkerler/oppo_ozip_decrypt #33: Boot.

sigitreal, 2020-01-01:
[image]
Help me... Im try to decrypt  boot.img on oppo k3, but this result..,.  Why? Im decrypt .ozip stockrom k3 is success.

sigitreal, 2020-01-01:
Link stockrom oppo k3 http://downloads.oppo.com.s3.amazonaws.com/firmware/CPH1955%20%28K3%29/Mp4/CPH1955EX_11_OTA_0050_all_tafZl9hKq1So.ozip

bkerler, 2020-02-02:
if you use it this way : ./ozipdecrypt.py ../CPH1955EX_11_OTA_0050_all_tafZl9hKq1So.ozip
you will get a file ./ozipdecrypt.py ../CPH1955EX_11_OTA_0050_all_tafZl9hKq1So.zip which will contain all decrypted data.

closed 2020-02-02 by bkerler