bkerler/oppo_ozip_decrypt #39: Hi, Could you please add Oppo Realme C2 RMX1941 AES Key?

ghostinashell2020, 2020-01-28:
Hi, 

I am trying to extract those key, but couldnt find it anywhere.. could you please help me to find it? Thank you very much.

ghostinashell2020, 2020-01-28:
Error

Found correct AES key: acac1e13a72531ae4a1b22bb31c1cc22
Traceback (most recent call last):
  File "ozipdecrypt.py", line 157, in <module>
    main()
  File "ozipdecrypt.py", line 129, in main
    with open(wfilename,'wb') as wf:
FileNotFoundError: [Errno 2] No such file or directory: 'out\\temp\\boot.img'

aIecxs, 2020-01-28:
#10

ghostinashell2020, 2020-01-29:
Hi mate, 
Trying to decrypt it, the error is:

Found correct AES key: acac1e13a72531ae4a1b22bb31c1cc22
Traceback (most recent call last):
File "ozipdecrypt.py", line 157, in
main()
File "ozipdecrypt.py", line 129, in main
with open(wfilename,'wb') as wf:
FileNotFoundError: [Errno 2] No such file or directory: 'out\temp\boot.img'

aIecxs, 2020-02-01:
try older version 844e75d use linux

bkerler, 2020-02-01:
Issue should be fixed for windows with latest commit 290d880

aIecxs, 2020-02-09:
successfully decrypts, but still fails for unencrypted files without header they are inside too. it stops after fail and doesn't continue

bkerler, 2020-02-12:
Should be fixed with latest commit  95b69b1

ghostinashell2020, 2020-02-13:
F:\blablabla\Oppo_Download_Tool\oppo_ozip_decrypt-master\oppo_ozip_decrypt-master>python ozipdecrypt.py F:\blablabla\REALMEC2A21\A21_OTA\RMX1941EX_11_OTA_0210_all_WBbnkGFmFuPz_local.ozip
ozipdecrypt 1.0 (c) B.Kerler 2017-2020
Unknown image, please report an issue with image name !

bkerler, 2020-02-13:
Hi, issue should be fixed with commit 3b1c8d3

bkerler, 2020-03-07:
Closing as no more issues were confirmed.

closed 2020-03-07 by bkerler