bkerler/oppo_ozip_decrypt #40: CPH1803  OppoA3s & CPH1909 OppoA5s keys is correct! but... properly extract!

PhilCodeEx, 2020-02-01:
ok iwill share the AES keys CPH1803 OppoA3s and A5s
 i already figure out how this firmware decrypt and encrypt in this two oppo 

14C2CD6214CFDC2733AE81B22BC1462C =CPH1803 OppoA3s
1CA21E12271435AE331B81BBA7C14612 = CPH1909 OppoA5s

in oppo a3s ota.zip is a regular zip file but every file inside ota is encrypted which is in offset 0x50 and 0x4050 after i manage to decrypt the line it will reveal the correct binary data

for example in boot.img in CPH1803 OppoA3s
in offset 
0x50 4E 21 FE AA BF F0 56 32 B5 06 85 B7 6C 3E 65 5F
after decrypt the line reveal the 
0x50 41 4E 44 52 4F 49 44 21 3B 9F B0 00 00 80 00 80 ANDROID!

and in modem.img offset  
0x50 74 52 D8 7B 93 04 5B ED 9E 59 F4 75 4F D4 63 65
0x4050  19 2A 06 CC 6E 68 AD 9B AD 6C 74 CB FA 83 AC 4F

after decrypt 
0x50 EB 3C 90 4D 53 44 4F 53 35 2E 30 00 02 20 01 00  <MSDOS5.0
0x4050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
than means the aes key used was correct 

can you create a script to auto patch the encrypted offset to decrypt the specific line  and delete the the first 0x0 offset up to from 0x50 wich is the OPPOENCRYPT! upto the beginning of the encrypted offset. of each extracted file??

bkerler, 2020-02-01:
yes, I will have a look

bkerler, 2020-02-01:
Support is added with commit eac7bfa . Please test.

PhilCodeEx, 2020-02-02:
Thank you fow adding support for CPH1803 OppoA3s

i will test it soon i get back to you after that 
i appreciate your work!

PhilCodeEx, 2020-02-02:
> Support is added with commit [eac7bfa](https://github.com/bkerler/oppo_ozip_decrypt/commit/eac7bfa826e76bbb7bab952603cb362daeff283f) . Please test.

after test im using windows 10 with elevated command prompt,

ozipdecypt.py CPH1803EX_11_OTA_0240_all_y9VVWBYOdndL.zip --mode=2
ozipdecrypt 0.5 (c) B.Kerler 2017-2019
Extracting CPH1803EX_11_OTA_0240_all_y9VVWBYOdndL.zip
Found correct AES key: 14C2CD6214CFDC2733AE81B22BC1462C
Decrypting temp\boot.img
Decrypting temp\firmware-update\adspso.img
Decrypting temp\firmware-update\cmnlib64_30.img
Decrypting temp\firmware-update\cmnlib_30.img
Decrypting temp\firmware-update\devcfg.img
Decrypting temp\firmware-update\dpAp.img
Decrypting temp\firmware-update\DRIVER.img
Decrypting temp\firmware-update\dtbo.img
Decrypting temp\firmware-update\emmc_appsboot.img
Decrypting temp\firmware-update\emmc_fw.img
Decrypting temp\firmware-update\keymaster64.img
Decrypting temp\firmware-update\lksecapp.img
Decrypting temp\firmware-update\logo.img
Decrypting temp\firmware-update\mdtp.img
Decrypting temp\firmware-update\modem.img
Decrypting temp\firmware-update\rpm.img
Decrypting temp\firmware-update\sbl1.img
Decrypting temp\firmware-update\static_nvbk.img
Decrypting temp\firmware-update\tz.img
Decrypting temp\firmware-update\vbmeta.img
Traceback (most recent call last):
  File "C:\Python\Python36\lib\shutil.py", line 538, in move
    os.rename(src, real_dst)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'temp\\META-INF\\CERT.RSA' -> 'out\\CERT.RSA'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\philcodex\Desktop\cph1803\ozipdecypt.py", line 224, in <module>
    main()
  File "C:\Users\philcodex\Desktop\cph1803\ozipdecypt.py", line 214, in main
    shutil.move(rfilename, wfilename)
  File "C:\Python\Python36\lib\shutil.py", line 553, in move
    os.unlink(src)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'temp\\META-INF\\CERT.RSA'

and this is the image of out folder
[image]

bkerler, 2020-02-05:
Looks good to me. Seems like the cert.rsa file was opened, that's why it failed.

PhilCodeEx, 2020-02-05:
> Looks good to me. Seems like the cert.rsa file was opened, that's why it failed.

even i move it to other location i keep getting error on extracting 

PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'temp\META-INF\CERT.RSA'

no other process running Elevated Command Prompt only.
even i restart machine keeps getting that error after executing in command prompt

PhilCodeEx, 2020-02-05:
i am thinking maybe the script cant find offset 0x50 and 0x4050 in CERT.RSA to decode thats why the code stock on that file

bkerler, 2020-02-07:
Could be a windows issue. For me, it went through all files on linux

proganime1200, 2020-03-03:
how do i repack this files to flash through stock recovery??

nathanvdh, 2020-03-05:
Even testing that same file now: CPH1803EX_11_OTA_0240_all_y9VVWBYOdndL.zip.
I get 'unknown aes key'. Is something broken in a recent commit?

bkerler, 2020-03-07:
@proganime2020: you can't unless you have the rsa private key for signing the zip.

@nathanvdh: I will have a look

bkerler, 2020-03-07:
@nathanvdh No, it's working fine. I guess you missed the mode 2 option.
Anyways, I added a mode 2 autodetect, thus mode 2 is no longer needed.
Tested and working with commit 91c717d. Closing this issue.

closed 2020-03-07 by bkerler