bkerler/oppo_ozip_decrypt #41: Unknown AES key

siddhrsh, 2020-02-11:
I am trying to extract Realme C2 ozip to zip via ** --mode 2 command but i am getting unknown aes key reverse key from recovery first error first method works but i want to extract files in zip

bkerler, 2020-02-12:
Please try the latest version

siddhrsh, 2020-02-15:
> Please try the latest version

Tried with latest version same issue 
1st method working
2nd method gives Unknown AES key, reverse key from recovery first! error
can i extract file to twrp flashable zip directly because 1st method extract files in img format

bkerler, 2020-03-07:
Added fixes in commit 91c717d. Please retry and report if it works. If it fails, please refer to the non-working ozip (full filename and/or link)

tuyennn, 2020-03-24:
Here the log

`python ozipdecrypt.py .\RMX1801EX_11_OTA_0270_all_KvuheffUBmQD.ozip
Traceback (most recent call last):
  File "ozipdecrypt.py", line 20, in <module>
    from Crypto.Cipher import AES
ModuleNotFoundError: No module named 'Crypto'`

siddhrsh, 2020-03-26:
> Added fixes in commit [91c717d](https://github.com/bkerler/oppo_ozip_decrypt/commit/91c717d6a6a5925a193c269e3b637f51b694ef4c). Please retry and report if it works. If it fails, please refer to the non-working ozip (full filename and/or link)
Thanks
its works but it extracted into img files is it possible to extract them in zip format

bkerler, 2021-01-24:
Should be fixed in commit 8cc89cd

closed 2021-01-25 by siddhrsh