bkerler/oppo_ozip_decrypt #43: Please help me find Oppo A71 2018 CPH1801 AES Key

aricu, 2020-02-25:
So i've been following tutorial for aes key extraction and i find all key from libapplypatch_jni.so (stock rom), here are the key that i find :

d6dccf0ad5acd4e0292e522db7c1381e  (mnkey)
d6eecf0ae5acd4e0e9fe522de7ce381e    (mkey)
d6eccf0ae5acd4e0e92e522de7c1381e   (userkey)
12341eaac4c123ce193556a1bbcc232d  (realkey)
d7dcce1ad4afdce2393e5161cbdc4321   (testkey)
d6dccf0ad5acd4e0292e522db7c1381e   (utilkey)

Then i fit all key that i find to ozipdecrypt script but i get "Unknown AES key, reverse key from recovery first!" result, i try to extract the key from libapplypatch_jni.so (ofp file) but the result is same. 
Please help me find the correct aes key

describe19, 2020-03-05:
It seems the script always fails in a recent commit. I think reverting this line in the keytest function inside the script fixes it:
Working line:
elif (dat[0:4] == b'\x41\x4E\x44\x52'):
Not working line:
elif (dat[0:4] == b'\x41\x56\x42\x30'):

bkerler, 2020-03-07:
I will have a look for it and fix it very soon. Thanks @describe19

bkerler, 2020-03-07:
Fixed and confirmed to work with commit 91c717d.

closed 2020-03-07 by bkerler