bkerler/oppo_ozip_decrypt #45: Can't Decrypting All Files

aricu, 2020-03-08:
I've found the right aes key for Oppo A71 2018 CPH1801 with this commit https://github.com/bkerler/oppo_ozip_decrypt/commit/91c717d6a6a5925a193c269e3b637f51b694ef4c, but i got several error like this :

Traceback (most recent call last):
  File "ozipdecrypt.py", line 211, in main
    if zo.extract('oppo_metadata', outpath):
  File "/usr/lib/python3.8/zipfile.py", line 1628, in extract
    return self._extract_member(member, path, pwd)
  File "/usr/lib/python3.8/zipfile.py", line 1667, in _extract_member
    member = self.getinfo(member)
  File "/usr/lib/python3.8/zipfile.py", line 1439, in getinfo
    raise KeyError(
KeyError: "There is no item named 'oppo_metadata' in the archive"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/shutil.py", line 781, in move
    os.rename(src, real_dst)
OSError: [Errno 26] Text file busy: 'temp/META-INF/CERT.RSA' -> 'out/CERT.RSA'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "ozipdecrypt.py", line 257, in <module>
    main()
  File "ozipdecrypt.py", line 217, in main
    mode2(filename)
  File "ozipdecrypt.py", line 161, in mode2
    shutil.move(rfilename, wfilename)
  File "/usr/lib/python3.8/shutil.py", line 796, in move
    os.unlink(src)
OSError: [Errno 26] Text file busy: 'temp/META-INF/CERT.RSA'

bkerler, 2021-01-24:
Can you please provide a link to that ozip file ?

bkerler, 2021-01-24:
Should be fixed in commit 8cc89cd

closed 2021-03-20 by bkerler