bkerler/oppo_ozip_decrypt #61: oppo A5: "There is no item named 'oppo_metadata' in the archive"

kwaiching, 2021-01-14:
`$ python3 ozipdecrypt.py '/media/kwaiching/9EC1B7A5700DC5DD/Android/AndroidRom/OPPO/OppoA5_q_450B/01Rom/tmp/PBBM30_11_OTA_0270_all_YfsPyB6OXxjg.ozip' 
ozipdecrypt 1.1 (c) B.Kerler 2017-2020
Detected mode 2....
Extracting /media/kwaiching/9EC1B7A5700DC5DD/Android/AndroidRom/OPPO/OppoA5_q_450B/01Rom/tmp/PBBM30_11_OTA_0270_all_YfsPyB6OXxjg.ozip
Traceback (most recent call last):
  File "ozipdecrypt.py", line 245, in main
    if zo.extract('oppo_metadata', outpath):
  File "/usr/lib/python3.8/zipfile.py", line 1630, in extract
    return self._extract_member(member, path, pwd)
  File "/usr/lib/python3.8/zipfile.py", line 1669, in _extract_member
    member = self.getinfo(member)
  File "/usr/lib/python3.8/zipfile.py", line 1441, in getinfo
    raise KeyError(
KeyError: "There is no item named 'oppo_metadata' in the archive"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "ozipdecrypt.py", line 292, in <module>
    main()
  File "ozipdecrypt.py", line 251, in main
    mode2(filename)
  File "ozipdecrypt.py", line 168, in mode2
    zipObj.extractall(temp)
  File "/usr/lib/python3.8/zipfile.py", line 1647, in extractall
    self._extract_member(zipinfo, path, pwd)
  File "/usr/lib/python3.8/zipfile.py", line 1701, in _extract_member
    open(targetpath, "wb") as target:
OSError: [Errno 22] Invalid argument: '/media/kwaiching/9EC1B7A5700DC5DD/Android/AndroidRom/OPPO/OppoA5_q_450B/01Rom/tmp/temp/vendor/firmware/emmc/HYNIX-hB8aP>.bin_crc.bin'
`

bkerler, 2021-01-24:
Should be fixed in commit 8cc89cd

kwaiching, 2021-01-27:
Thanks a lot

closed 2021-03-20 by bkerler

Shivam-0-0, 2023-05-18:
plzz help me i didnt understand it how to fix that?

Kalia00739, 2023-11-17:
C:\Users\Misbah Ullah\AppData\Local\Programs\Python\Python312>python ozipdecrypt.py R11.ozip
ozipdecrypt 1.32 (c) B.Kerler 2017-2022
"There is no item named 'oppo_metadata' in the archive"
Detected mode 2....
C:\Users\Misbah Ullah\AppData\Local\Programs\Python\Python312\ozipdecrypt.py:87: DeprecationWarning: onerror argument is deprecated, use onexc instead
  shutil.rmtree(path, onerror=del_rw)
Finding key...  R11.ozip
Unknown mode2, boot.img wasn't encrypted
Extracting...  R11.ozip
Traceback (most recent call last):
  File "C:\Users\Misbah Ullah\AppData\Local\Programs\Python\Python312\ozipdecrypt.py", line 258, in main
    if zo.extract('oppo_metadata', outpath):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Misbah Ullah\AppData\Local\Programs\Python\Python312\Lib\zipfile\__init__.py", line 1703, in extract
    return self._extract_member(member, path, pwd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Misbah Ullah\AppData\Local\Programs\Python\Python312\Lib\zipfile\__init__.py", line 1742, in _extract_member
    member = self.getinfo(member)
             ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Misbah Ullah\AppData\Local\Programs\Python\Python312\Lib\zipfile\__init__.py", line 1515, in getinfo
    raise KeyError(
KeyError: "There is no item named 'oppo_metadata' in the archive"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Misbah Ullah\AppData\Local\Programs\Python\Python312\ozipdecrypt.py", line 318, in <module>
    sys.exit(main(args.filename))
             ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Misbah Ullah\AppData\Local\Programs\Python\Python312\ozipdecrypt.py", line 265, in main
    return mode2(filename)
           ^^^^^^^^^^^^^^^
  File "C:\Users\Misbah Ullah\AppData\Local\Programs\Python\Python312\ozipdecrypt.py", line 208, in mode2
    os.remove(os.path.join(temp,"out"))
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\Misbah Ullah\\AppData\\Local\\Programs\\Python\\Python312\\temp\\out'

Kalia00739, 2023-11-17:
please help me