bkerler/oppo_ozip_decrypt #64: mode2 extraction partially broken after commit 8cc89cd

rokibhasansagar, 2021-01-30:
### After https://github.com/bkerler/oppo_ozip_decrypt/commit/8cc89cd299a25eb85a60cdf562d1d0a19ce4cc0c, the ozip files which are not encrypted (actually just plain zip files) are not extracting correctly.

- I tested two OTAs for RMX2185, [RMX2185_11_OTA_0790](https://download.c.realme.com/osupdate/RMX2185_11_OTA_0790_all_dsysIPFhgWAX.ozip) and [RMX2185_11_OTA_0810](https://download.c.realme.com/osupdate/RMX2185_11_OTA_0810_all_jLGoLXnVv2Zm.ozip)
- Both fail with latest update, but works fine till https://github.com/bkerler/oppo_ozip_decrypt/commit/d02128dade8fffaf16e00f9f7d01f7be39558f69

### But _HEAD_ commit works fine with other encrypted ozips
- I tested [RMX2121_11_OTA_0180](https://download.c.realme.com/osupdate/RMX2121_11_OTA_0180_all_MoeO6C0vgRhB.ozip) and it de-crypted perfectly

bkerler, 2021-03-20:
I'll have a look.

bkerler, 2021-03-20:
Fixed with commit 1a1ce49

closed 2021-03-20 by bkerler