bkerler/oppo_ozip_decrypt #70: Unknown image - CPH2001PUEX_11_OTA_1470_all_GlioAPybO0Vk.ozip

c0bnut, 2021-04-12:
Device: Oppo A91
Image name: CPH2001PUEX_11_OTA_1470_all_GlioAPybO0Vk.ozip
Firmware download location: https://support.oppo.com/au/software-update/software-download/?m=A91

c0bnut, 2021-04-14:
Probably no point doing this now as Oppo has restricted access to the bootloader. But if you're feeling like a challenge you might be able to reverse engineer the "Oppo Deep Test APK" and unlock the bootloader somehow :) https://community.coloros.com/thread-50668-1-1.html

c0bnut, 2021-04-18:
Is there any possibility of being able to encrypt a zip back into ozip so it can be flashed with recovery? This would allow us to bypass beeding to unlock fastboot.

bkerler, 2021-08-07:
To put it simply, no, this won't work.

yssreddy1961, 2021-08-24:
i have also tested above CPH2021 i am also getting unknown image format .pl solve
with thanks

yssreddy1961, 2021-08-24:
OPPOENCRYPT!^@^@^@^@26656^@^@^@^@^@^@^@^@^@^@^@5b5205f28725ec790f98c8309a37833930f4ce12^@^@^@^@^@^@^@^@android_version=10
google_patch=20210405
os_version=V7.1
ota-downgrade=no
ota-id=CPH2001PUEX_11.C.53_1530_202104010441
ota-property-files=metadata:69:344
ota-required-cache=0
ota-type=BLOCK
ota_version=CPH2001PUEX_11.C.53_1530_202104010441
patch_type=1
post-timestamp=1617226813
pre-device=OP4B9B
version_name=CPH2001EX_11_C.53
wipe=0
^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@>
JW3]ٷ�R�kD��^R�q���ņ�_q.�q�6�%<��^YJ����:��J0^En^O^D�Qu��OB��75|�c�^\ǝ�nN�^D���F�q!��M^R��^Z �L�^B:�^F6�^_      �����~�8"�H�^LGo�D���O8^V#�c
��~�bN��e��[^W/��^W^FؠY��ǘ��d^Z�C�H�G��1�%���ն~H�^�,<^V���^NZ^H%��B�.��^U�s����^K{^F5�Z�^Mֆ.�;)Oa[A^F63_��P{2^Wn5+�^G�9^M�u��6��nT��V_j�y�j|�^Z��I^K���0]���T��Su�Xv_�^R�L^MT1a^H/�T�^^��])>���D^N>���z֡`�2:�1�>
���l��쌴���w��^@�^X���^Z�}^Q�6+���#;����^H+܀�^D
W*{�\�^]�^@M�^`57�q�@�^]��MU^T@?��      �x[��ܝM�8��6*^Drh�N�G?%�������=7^V����^L�^Qr�0;^D�EjG^F���hl41�|
�(��:�缒�^C^P��"0�!h7ߖΧ+5�ュm��B�W�I�ۋ^K\^@�#ɒ^X�O~��)���Ɉ�cj���uuf����^Ab�ޘ^G�qϖ ^D^L�y�^]��/H_2�)1-�^r^V�ET5v�^C�%^^Ht�R<^D�l^SīH�^A^S~�`�^S^[�(>�^S5�w.H��+Ɓ�C\Q^FԎ�F�/^M^Xĉ���Lʹx^D^E�^H.$^L;^O
^]�ݙ]��%�.�Q�L7�^?      (ٵ��t\VD^QTWS^^�k��O%�^]��_�-H4d�ȷ��r��D^V�ҚR~H��3^Tܶ^F^T&^MR&�Pg���6GI^Yc]������ܻ��R^_��^A�a�`�^V^^�^^!x{2?�z��˄��d���oqn�.c�po^F�ɗ^P^FM�S&�4�^H�˼^K�7?^Y�^UDv^]X�n�����)Qi�^TR^M��YmU9>
�D^]sr��^_�\��߇�J"5荶^Q^N��,���i�y(^R����� ���^W��!�"Dάp���J�y[�e^�������"��z6�؟�^MQ^GlF�b�^R^N�^D$;gh��[��HPj^XW�yz�8��t^K2^S�^LR     "��^R��-^A�Fw%�5%ߐ�Ic��^B@�XI\��O��^R��<�(�e���^Dc|u�)S^V@R���Y�#���V�>
�p�
5��fT�N��j����^Vv^D��^B1� H^H9
+f�     ����^��Qb���^Y|^O^_��DF�^Pi�g>�bh�XJئ^]6:�^V֥�� 5��(�^QK����W�t6V��N�LV�^@˱����j��ޓ^\?�^F��-��^N^T7W^Ldi�K�oJ�   �[ք%9^H� ӻ�[^V,��1I>���lˍ��+��+G+^D��^B-�^C^P{^Q�L�^V�^D�N^K���^X�!$���Yͅ��:Zg��^N�J��3>
^X�l^R#���k*���������ȑ��`�^B:�^�^P^C����W�/����gnJ�9�tI� g-���vF�<s^L��^S^?�b^O��M}g��&����^V�gT����]���^W�c8|�^D��͙�S�B{��^M��d^U���IӼ1i6͆x�������]e^@�P^M������^R�j_^^�^_H&�b���Z�^Cچ���- t�C�^MSu�$^G�\2�vN��>
����9^W��z�V=�W�]㝡��^?��Pg3 ��f$`��^Nu�nK�z�C^Z^S�`�V�%^_~�WQ�����s�&u�����F�� S�G[��^Fh ��;��^L^C�ai��.@���R^P��p     ��^�^S�1^H�������}^?Lo�3�6^U^@^T�^^"0dg@Ɇ%=GQ�^?�A�I���^Ue.��^Pg�c^_��DED�Qx^Z'�~�)�<>
D^Z
�D�M5a
P^Rw^O�m}�jk.�
^@�]^^a^E�r�^L��^F��\�r�^]��^B�j��Z�AxT�%�fȈ^P�ݏ���F�����!��n�3���HE�Ʋfd�F^B|�[A�^E^[   �N�^^^SΉ��3�m"�^@V,Cs!l&/�^F:-��W^Z^B���h�^_�+�r(5��^N�����/�H��^BI]d^L7�ƭ!B^H3��^U>^O<�V^SV�c  p�X��^EI/^@x��&��O�x>#>
4!qh^Q��'�:^_�^Bc$�^@Ni���^[�[Ǽ^N�n^X:���2lY8h�ּs��n<�S^H��^Gu�,�$�G��^Z±�.��q^P�P��^Ac)RH^L`6Z�^Q���Bں?^S7ѷx,u
"A+v��*���'��P�?.��������ῴlL6�6_J۸^Vw��j�-^Yc^`O��^^�A
�^Si�j�"�=k�@頢>)jU^Z?���s;�^_8���W/�>���^K.^C^L`�ͽx!��b�1���D/�^T���^K^O\~jS���        !?U�4������^@^Et���>�^D^X?�^[��I�^M�^@�F^Xz^_�]�z�^N�g����L:hef��^WuW��+/�r������)[n`��z^O_^B�@�^VMZ�ma��R�����^^'�uH^>
J�e�«��^P       3N̨���0D^\��}�X���O^T_���c�����vS�le��%6^H+��=di�e^C��z)�eI=���$|��3�1�R^O�A�^_�Zı^R^B��^4ߟ掐�dV^K�J�^@��ࠪ^]^K7QuGа^SM`.{s܃z9�z7��^T��@�~I�����ŋ� P���u$24�5�u�Ĩ�|^T�օ^[^O�\���47^�^GP��tP;�z`)�>
\P-Qc�^B)e �^H�(&�}^Xw��+^D���^BO�˱�E.^N��г{�~�!FkY&�*��^?����^GLx�^F/�o�^PjW^U�%`3�$!j�tjݞ<f�s�Ȑ)�^HƗ�f^a��J�Mkt�/L5%r]��^K�E�^C�vdx��1�@�?p��^_.�g\]�H^C^?���M[���<�<��^U5($`�^P�^?�V�^G�^F��_a�S��q��-�(�]B>
k���iA^[�U[�^V8??��~��: �1��^Z��^F�Ιϟ^_�g��$��{�φoZ��^\�մ!]W��Ϟ��C+�xR�^@*\��ó�'ׂ�T�^L^^9�^Y�&^O�j��^Eޛ^U���_�</ۧ�^NM�^_�q�^Z��A^H���^L�^Rj,���m�H^U~79qQ�t,�S�24t��*��/1@j|�^Y*���^F=�j^VMR�:,$�^O��^@ ��^HQ��2>
^Tl^���K^AC^C�"�^L8�d�\^E �B{�^M^C����=�Q����t^@".��]�����b�����{'�X��^Z^]�^W9�Ps{�z��^[)^A^X?�j@�d�Y���C�^A"��M��8}�S��ڲ3�P�n��t^T��2�ⶫ����������Hs���O7���?a�p����9^Y^C;��ү��{c^_
1>��N^G54^MҰ�D#��j�^O^Gm>��;^P��!ٛ�^[^Q�*������::i�2�^H���&�^Gμ���^L�^]V*�e����|K^B �WY��G��2e�¿� ����}�~b7      UYk�^\���j�&�ߚ���6p^Vwr�^DL��^\$Q�}k�^Q�^H�ٮ̶���[^Q,��i�^C�f�}�K�-Rk<�gM����^L�v���aΆt��lU,W���>
H���;^Z~ܯ�(���^A��Nt�v^V1�(GD<Hٸ�^Q�8�o
|np��Z�ۥ�R�Xmf;D�J��~^K�*�V}h�(>��3������sd!���I|�/^^������^Su�"��^D���0��\�^B^S��s2���k^E�^DTi^W�^Ua
j���ܔ��^F���W���y���ky^M<���^V^C�^Mq��G^\�3zX\�jcӺ^Qc.$--��0��^\^Y�������N��4�q ܲ�=>��>�!**8n�S���:���S(p|�^^���ys�h��I�[���=RH�3^M^LlT�^^^P�^_e^A���L5�^S?~E    [���>�0F-�c�h�̕��NT^L�+^FW�d^^x�=Ak�$^Q�Z��_��^>
����ɷb��(�]%�
<_>��5^P^K\�    �5�^^^_��$�`��)4:]�~�� *��Y��PI��Gn�POw��53^N��^_�o�.�K0��^\W��m�^\^Mv��P�^M@�b�ѯJ�^]�v�þ-��f�^T^XS����jx���7����i=�@�^L��%J�^E���� ȁR�^K<9W�V|ه{�^O(�
��u"���m^TI'h����^EWHj��F��;^A�]�P^XV����V��H�^P�.w�f���^_�(^K���Z�)��C���m>%�^B        �8��/^^j}\)��3ve���\^m�j*^H^?������^[~ �Ᏽ�&O��l^G��O �-�?��Wҡ���r�Z<�^C��HI���&���.�^A�
j}s���^?@'\����+�l^XF�eh�^WE�rODs��Fge���r�%�q;^\�B��u��4p{�[�&�^H^R�$��^@^]�&˹��:ջ�Q�k^F�-Q�#Y٧�m�O����^@^?D���8�8Z�S��f�j%�5�+��^_W�a^O`(䮹��1 Ur�%��1I�~�������A�u���kLxt���^H��C�M.b(^M�sW:�",^T,c;�������>
�^SW^\y)�Lmxi3sZ-Ү-#��hN&�I^_L���m-����O9�h^P�^^�^A�-Z���G�^O�ҵ�":R��^L��>^W�^A&�:^L�^W2��\*uא�v=V^Y�㐸keƋ^G^A��^G^Z^U����N.�^Rعfla�0�^G�ۊ�^@[
^Y�^M�^K6(^?��I���^Z^F^_^T�����^M�~Ѫ��^S)^X^T�^D��_���&�p!E��? ��B/�Z�iaHZ�@�柔 �樣�G���]A����f�*A�S�^^^R[X}-D��C�~���桊�^?�Y^U����m^B<���bX�ԿI�^Ay���n���s�S�A�V����G^L
t���^L��^StK�ise?&�N�(�)2�#�6@^X��K��J^P�d�^\�^D���hb��C�^B�0����vMS����e^HMa+jH^\�$v�\�1���[���^A�dr�^?���ZA%R��dk�M��A�Ɗ�^Odw�ZZa��v��I���{3�M�       g|o�E*7^P�ԥ�k��J�^H5���5$�j�DM�Q��F0�+�{-^?��̓��^T�q>��>
[... cut ...]
^C��D�^\^_�b��^E^Q���\�>��<^G��A\�w^[�

^G Help          ^O Write Out     ^W Where Is      ^K Cut           ^T Execute       ^C Location      M-U Undo         M-A Set Mark     M-] To Bracket   M-Q Previous     ^B Back          ^◀ Prev Word
^X Exit          ^R Read File     ^\ Replace       ^U Paste         ^J Justify       ^_ Go To Line    M-E Redo         M-6 Copy         ^Q Where Was     M-W Next         ^F Forward       ^▶ Next Word

pl decrypt  system.transfer.list extracted from above ozip file.boot.img is also like that

yssreddy1961, 2021-08-24:
OPPOENCRYPT!^@^@^@^@26656^@^@^@^@^@^@^@^@^@^@^@5b5205f28725ec790f98c8309a37833930f4ce12^@^@^@^@^@^@^@^@android_version=10
google_patch=20210405
os_version=V7.1
ota-downgrade=no
ota-id=CPH2001PUEX_11.C.53_1530_202104010441
ota-property-files=metadata:69:344
ota-required-cache=0
ota-type=BLOCK
ota_version=CPH2001PUEX_11.C.53_1530_202104010441
patch_type=1
post-timestamp=1617226813
pre-device=OP4B9B
version_name=CPH2001EX_11_C.53
wipe=0
^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@>
JW3]ٷ�R�kD��^R�q���ņ�_q.�q�6�%<��^YJ����:��J0^En^O^D�Qu��OB��75|�c�^\ǝ�nN�^D���F�q!��M^R��^Z �L�^B:�^F6�^_      �����~�8"�H�^LGo�D���O8^V#�c
��~�bN��e��[^W/��^W^FؠY��ǘ��d^Z�C�H�G��1�%���ն~H�^�,<^V���^NZ^H%��B�.��^U�s����^K{^F5�Z�^Mֆ.�;)Oa[A^F63_��P{2^Wn5+�^G�9^M�u��6��nT��V_j�y�j|�^Z��I^K���0]���T��Su�Xv_�^R�L^MT1a^H/�T�^^��])>���D^N>���z֡`�2:�1�>
���l��쌴���w��^@�^X���^Z�}^Q�6+���#;����^H+܀�^D
W*{�\�^]�^@M�^`57�q�@�^]��MU^T@?��      �x[��ܝM�8��6*^Drh�N�G?%�������=7^V����^L�^Qr�0;^D�EjG^F���hl41�|
�(��:�缒�^C^P��"0�!h7ߖΧ+5�ュm��B�W�I�ۋ^K\^@�#ɒ^X�O~��)���Ɉ�cj���uuf����^Ab�ޘ^G�qϖ ^D^L�y�^]��/H_2�)1-�^r^V�ET5v�^C�%^^Ht�R<^D�l^SīH�^A^S~�`�^S^[�(>�^S5�w.H��+Ɓ�C\Q^FԎ�F�/^M^Xĉ���Lʹx^D^E�^H.$^L;^O
^]�ݙ]��%�.�Q�L7�^?      (ٵ��t\VD^QTWS^^�k��O%�^]��_�-H4d�ȷ��r��D^V�ҚR~H��3^Tܶ^F^T&^MR&�Pg���6GI^Yc]������ܻ��R^_��^A�a�`�^V^^�^^!x{2?�z��˄��d���oqn�.c�po^F�ɗ^P^FM�S&�4�^H�˼^K�7?^Y�^UDv^]X�n�����)Qi�^TR^M��YmU9>
�D^]sr��^_�\��߇�J"5荶^Q^N��,���i�y(^R����� ���^W��!�"Dάp���J�y[�e^�������"��z6�؟�^MQ^GlF�b�^R^N�^D$;gh��[��HPj^XW�yz�8��t^K2^S�^LR     "��^R��-^A�Fw%�5%ߐ�Ic��^B@�XI\��O��^R��<�(�e���^Dc|u�)S^V@R���Y�#���V�>
�p�
5��fT�N��j����^Vv^D��^B1� H^H9
+f�     ����^��Qb���^Y|^O^_��DF�^Pi�g>�bh�XJئ^]6:�^V֥�� 5��(�^QK����W�t6V��N�LV�^@˱����j��ޓ^\?�^F��-��^N^T7W^Ldi�K�oJ�   �[ք%9^H� ӻ�[^V,��1I>���lˍ��+��+G+^D��^B-�^C^P{^Q�L�^V�^D�N^K���^X�!$���Yͅ��:Zg��^N�J��3>
^X�l^R#���k*���������ȑ��`�^B:�^�^P^C����W�/����gnJ�9�tI� g-���vF�<s^L��^S^?�b^O��M}g��&����^V�gT����]���^W�c8|�^D��͙�S�B{��^M��d^U���IӼ1i6͆x�������]e^@�P^M������^R�j_^^�^_H&�b���Z�^Cچ���- t�C�^MSu�$^G�\2�vN��>
����9^W��z�V=�W�]㝡��^?��Pg3 ��f$`��^Nu�nK�z�C^Z^S�`�V�%^_~�WQ�����s�&u�����F�� S�G[��^Fh ��;��^L^C�ai��.@���R^P��p     ��^�^S�1^H�������}^?Lo�3�6^U^@^T�^^"0dg@Ɇ%=GQ�^?�A�I���^Ue.��^Pg�c^_��DED�Qx^Z'�~�)�<>
D^Z
�D�M5a
P^Rw^O�m}�jk.�
^@�]^^a^E�r�^L��^F��\�r�^]��^B�j��Z�AxT�%�fȈ^P�ݏ���F�����!��n�3���HE�Ʋfd�F^B|�[A�^E^[   �N�^^^SΉ��3�m"�^@V,Cs!l&/�^F:-��W^Z^B���h�^_�+�r(5��^N�����/�H��^BI]d^L7�ƭ!B^H3��^U>^O<�V^SV�c  p�X��^EI/^@x��&��O�x>#>
4!qh^Q��'�:^_�^Bc$�^@Ni���^[�[Ǽ^N�n^X:���2lY8h�ּs��n<�S^H��^Gu�,�$�G��^Z±�.��q^P�P��^Ac)RH^L`6Z�^Q���Bں?^S7ѷx,u
"A+v��*���'��P�?.��������ῴlL6�6_J۸^Vw��j�-^Yc^`O��^^�A
�^Si�j�"�=k�@頢>)jU^Z?���s;�^_8���W/�>���^K.^C^L`�ͽx!��b�1���D/�^T���^K^O\~jS���        !?U�4������^@^Et���>�^D^X?�^[��I�^M�^@�F^Xz^_�]�z�^N�g����L:hef��^WuW��+/�r������)[n`��z^O_^B�@�^VMZ�ma��R�����^^'�uH^>
J�e�«��^P       3N̨���0D^\��}�X���O^T_���c�����vS�le��%6^H+��=di�e^C��z)�eI=���$|��3�1�R^O�A�^_�Zı^R^B��^4ߟ掐�dV^K�J�^@��ࠪ^]^K7QuGа^SM`.{s܃z9�z7��^T��@�~I�����ŋ� P���u$24�5�u�Ĩ�|^T�օ^[^O�\���47^�^GP��tP;�z`)�>
\P-Qc�^B)e �^H�(&�}^Xw��+^D���^BO�˱�E.^N��г{�~�!FkY&�*��^?����^GLx�^F/�o�^PjW^U�%`3�$!j�tjݞ<f�s�Ȑ)�^HƗ�f^a��J�Mkt�/L5%r]��^K�E�^C�vdx��1�@�?p��^_.�g\]�H^C^?���M[���<�<��^U5($`�^P�^?�V�^G�^F��_a�S��q��-�(�]B>
k���iA^[�U[�^V8??��~��: �1��^Z��^F�Ιϟ^_�g��$��{�φoZ��^\�մ!]W��Ϟ��C+�xR�^@*\��ó�'ׂ�T�^L^^9�^Y�&^O�j��^Eޛ^U���_�</ۧ�^NM�^_�q�^Z��A^H���^L�^Rj,���m�H^U~79qQ�t,�S�24t��*��/1@j|�^Y*���^F=�j^VMR�:,$�^O��^@ ��^HQ��2>
^Tl^���K^AC^C�"�^L8�d�\^E �B{�^M^C����=�Q����t^@".��]�����b�����{'�X��^Z^]�^W9�Ps{�z��^[)^A^X?�j@�d�Y���C�^A"��M��8}�S��ڲ3�P�n��t^T��2�ⶫ����������Hs���O7���?a�p����9^Y^C;��ү��{c^_
1>��N^G54^MҰ�D#��j�^O^Gm>��;^P��!ٛ�^[^Q�*������::i�2�^H���&�^Gμ���^L�^]V*�e����|K^B �WY��G��2e�¿� ����}�~b7      UYk�^\���j�&�ߚ���6p^Vwr�^DL��^\$Q�}k�^Q�^H�ٮ̶���[^Q,��i�^C�f�}�K�-Rk<�gM����^L�v���aΆt��lU,W���>
H���;^Z~ܯ�(���^A��Nt�v^V1�(GD<Hٸ�^Q�8�o
|np��Z�ۥ�R�Xmf;D�J��~^K�*�V}h�(>��3������sd!���I|�/^^������^Su�"��^D���0��\�^B^S��s2���k^E�^DTi^W�^Ua
j���ܔ��^F���W���y���ky^M<���^V^C�^Mq��G^\�3zX\�jcӺ^Qc.$--��0��^\^Y�������N��4�q ܲ�=>��>�!**8n�S���:���S(p|�^^���ys�h��I�[���=RH�3^M^LlT�^^^P�^_e^A���L5�^S?~E    [���>�0F-�c�h�̕��NT^L�+^FW�d^^x�=Ak�$^Q�Z��_��^>
����ɷb��(�]%�
<_>��5^P^K\�    �5�^^^_��$�`��)4:]�~�� *��Y��PI��Gn�POw��53^N��^_�o�.�K0��^\W��m�^\^Mv��P�^M@�b�ѯJ�^]�v�þ-��f�^T^XS����jx���7����i=�@�^L��%J�^E���� ȁR�^K<9W�V|ه{�^O(�
��u"���m^TI'h����^EWHj��F��;^A�]�P^XV����V��H�^P�.w�f���^_�(^K���Z�)��C���m>%�^B        �8��/^^j}\)��3ve���\^m�j*^H^?������^[~ �Ᏽ�&O��l^G��O �-�?��Wҡ���r�Z<�^C��HI���&���.�^A�
j}s���^?@'\����+�l^XF�eh�^WE�rODs��Fge���r�%�q;^\�B��u��4p{�[�&�^H^R�$��^@^]�&˹��:ջ�Q�k^F�-Q�#Y٧�m�O����^@^?D���8�8Z�S��f�j%�5�+��^_W�a^O`(䮹��1 Ur�%��1I�~�������A�u���kLxt���^H��C�M.b(^M�sW:�",^T,c;�������>
�^SW^\y)�Lmxi3sZ-Ү-#��hN&�I^_L���m-����O9�h^P�^^�^A�-Z���G�^O�ҵ�":R��^L��>^W�^A&�:^L�^W2��\*uא�v=V^Y�㐸keƋ^G^A��^G^Z^U����N.�^Rعfla�0�^G�ۊ�^@[
^Y�^M�^K6(^?��I���^Z^F^_^T�����^M�~Ѫ��^S)^X^T�^D��_���&�p!E��? ��B/�Z�iaHZ�@�柔 �樣�G���]A����f�*A�S�^^^R[X}-D��C�~���桊�^?�Y^U����m^B<���bX�ԿI�^Ay���n���s�S�A�V����G^L
t���^L��^StK�ise?&�N�(�)2�#�6@^X��K��J^P�d�^\�^D���hb��C�^B�0����vMS����e^HMa+jH^\�$v�\�1���[���^A�dr�^?���ZA%R��dk�M��A�Ɗ�^Odw�ZZa��v��I���{3�M�       g|o�E*7^P�ԥ�k��J�^H5���5$�j�DM�Q��F0�+�{-^?��̓��^T�q>��>
[... cut ...]
��I^]ND��zv�^X�Q&^B���^R�X'�k���^T^OR^Us!^2ߣ:�$@�9�����.^G�E�$|B� �*^U�v�^X"GI��!A����s^K�^\�`�ˣ��^[�U��5���DS�|�����\A�1���:�逰,A{^?��^@F����^Bt�si�^@.)"�Z-�a�ȋժ4�e���ߐ^D���۱O/^L��8X�h�Y��r^Te^N9����\^A�~�>
^C��D�^\^_�b��^E^Q���\�>��<^G��A\�w^[�

^G Help          ^O Write Out     ^W Where Is      ^K Cut           ^T Execute       ^C Location      M-U Undo         M-A Set Mark     M-] To Bracket   M-Q Previous     ^B Back          ^◀ Prev Word
^X Exit          ^R Read File     ^\ Replace       ^U Paste         ^J Justify       ^_ Go To Line    M-E Redo         M-6 Copy         ^Q Where Was     M-W Next         ^F Forward       ^▶ Next Word
boot.img file

LoveCambodia, 2021-09-08:
lcambodia166@gmail.com

closed 2022-11-07 by bkerler