bkerler/oppo_ozip_decrypt #80: No able to decrypt file REALME 7 RMX2151

RobertST25, 2022-02-28:
Hello there,

Sorry to bother you, I am newbie at this program and was wondering if you could help me to solve my problem.

I am not able to decrypt ozip file. No matter how I try, always error.
I will have to learn about that, but for now, if you would be so nice I would appreciate your help, please see CMD output:
¨
Thank you very much,
Robert

AppData\Local\Programs\Python\Python39>python ozipdecrypt.py realme.ozip
ozipdecrypt 1.31 (c) B.Kerler 2017-2021
"There is no item named 'oppo_metadata' in the archive"
Detected mode 2....
Finding key...  realme.ozip
Unknown mode2, boot.img wasn't encrypted
Extracting...  realme.ozip
Traceback (most recent call last):
  File "C:\Users\stehl\AppData\Local\Programs\Python\Python39\ozipdecrypt.py", line 258, in main
    if zo.extract('oppo_metadata', outpath):
  File "C:\Users\stehl\AppData\Local\Programs\Python\Python39\lib\zipfile.py", line 1616, in extract
    return self._extract_member(member, path, pwd)
  File "C:\Users\stehl\AppData\Local\Programs\Python\Python39\lib\zipfile.py", line 1655, in _extract_member
    member = self.getinfo(member)
  File "C:\Users\stehl\AppData\Local\Programs\Python\Python39\lib\zipfile.py", line 1429, in getinfo
    raise KeyError(
KeyError: "There is no item named 'oppo_metadata' in the archive"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\stehl\AppData\Local\Programs\Python\Python39\ozipdecrypt.py", line 317, in <module>
    sys.exit(main(args.filename))
  File "C:\Users\stehl\AppData\Local\Programs\Python\Python39\ozipdecrypt.py", line 265, in main
    return mode2(filename)
  File "C:\Users\stehl\AppData\Local\Programs\Python\Python39\ozipdecrypt.py", line 208, in mode2
    os.remove(os.path.join(temp,"out"))
PermissionError: [WinError 32] Proces nemá přístup k souboru, neboť jej právě využívá jiný proces: 'C:\\Users\\stehl\\AppData\\Local\\Programs\\Python\\Python39\\temp\\out'

C:\Users\stehl\AppData\Local\Programs\Python\Python39>

kokare-darshan, 2022-04-10:
Even I am facing this issue

bkerler, 2022-11-07:
It's not encrypted. Closing this issue.

closed 2022-11-07 by bkerler