bkerler/oppo_ozip_decrypt #81: "There is no item named 'oppo_metadata' in the archive" OPPO A5S

Greednaught, 2022-04-24:
ozipdecrypt 1.31 (c) B.Kerler 2017-2021
"There is no item named 'oppo_metadata' in the archive"
Detected mode 2....
Finding key...  a5s.ozip
Found correct AES key: 1CA21E12271435AE331B81BBA7C14612
Extracting...  a5s.ozip
Traceback (most recent call last):
  File "C:\Users\Christian Pineda\Desktop\Python\ozipdecrypt.py", line 258, in main
    if zo.extract('oppo_metadata', outpath):
  File "C:\Users\Christian Pineda\Desktop\Python\lib\zipfile.py", line 1626, in extract
    return self._extract_member(member, path, pwd)
  File "C:\Users\Christian Pineda\Desktop\Python\lib\zipfile.py", line 1665, in _extract_member
    member = self.getinfo(member)
  File "C:\Users\Christian Pineda\Desktop\Python\lib\zipfile.py", line 1439, in getinfo
    raise KeyError(
KeyError: "There is no item named 'oppo_metadata' in the archive"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Christian Pineda\Desktop\Python\ozipdecrypt.py", line 317, in <module>
    sys.exit(main(args.filename))
  File "C:\Users\Christian Pineda\Desktop\Python\ozipdecrypt.py", line 265, in main
    return mode2(filename)
  File "C:\Users\Christian Pineda\Desktop\Python\ozipdecrypt.py", line 208, in mode2
    os.remove(os.path.join(temp,"out"))
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\Christian Pineda\\Desktop\\Python\\temp\\out'

Cybercanasa10, 2022-08-22:
how? what do you change

ZYTEric, 2022-11-04:
Changing ozipdecrypt.py line 258 to ' if zo.extractall(outpath) ' might be useful

TickledOnions, 2022-12-29:
> should be fixed with [1a1ce49](https://github.com/bkerler/oppo_ozip_decrypt/commit/1a1ce49f429096c3dcac740509d8488c899e03b1)

It is still not fixed for me

TickledOnions, 2023-01-03:
I found a way to fix this error:
Use this fork https://github.com/wzsx150/oppo_ozip_decrypt